ros-industrial/universal_robot #571: How can I limit velocity?

Ryoga723, 2021-05-12:
Hello,

When we use UR, we can limit its velocity by teach pendant.

But it's stupid to manipulate teach pendant every time.

So I'd like to change limitation in moveit program.

Now I use moveit::planning_interface::MoveGroupInterface move_group class and to execute move_group.execute(my plan) is method to move robot.

Thank you for your helping.

gavanderhoorn, 2021-05-12:
Which driver are you using?

Ryoga723, 2021-05-12:
I use ur_robot_driver. 

launch command is the same as read.me of this package. 
roslaunch ur_bringup ur5_bringup.launch robot_ip:=IP_OF_THE_ROBOT 

And I use below command to connect moveit.
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch

gavanderhoorn, 2021-05-12:
> I use ur_robot_driver.

Then please post this on [UniversalRobots/Universal_Robots_ROS_Driver/issues](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues).

> launch command is the same as read.me of this package.
> ```
> roslaunch ur_bringup ur5_bringup.launch robot_ip:=IP_OF_THE_ROBOT
> ```

that doesn't seem right: note you're referring to `ur_bringup`. The readme you refer to actually launch things from `ur_robot_driver`.

gavanderhoorn, 2021-05-12:
As this is not an issue with the package(s) in this repository, I'm going to close it.

closed 2021-05-12 by gavanderhoorn