ros-industrial/universal_robot #572: Strange MoveIt commander behavior with ROS Noetic

dudasdavid, 2021-05-19:
Hi Universal Robots team,

I know that there isn't any official support yet for Noetic in this repository, but I can see that you're working on it in the Universal_Robots_ROS_Driver. @fmauch I'm using a fork based on your calibration_devel branch, but I tested your calibration_devel-staging, too.

I have no issues with ROS Melodic, only with Noetic and only when using moveit commander with pose target (using set_pose_target).

You can reproduce my issue with the following steps:
1) `git clone -b calibration_devel-staging https://github.com/fmauch/universal_robot`
2) Change `arm_controller` to `pos_joint_traj_controller` in ur3e_moveit_planning_execution.launch
3) Launch the following launchfiles:
- `roslaunch ur_gazebo ur3e_bringup.launch limited:=true`
- `roslaunch ur3e_moveit_config ur3e_moveit_planning_execution.launch sim:=true limited:=true`
4) Run [my example code that can be found here](https://gist.github.com/dudasdavid/174816564d0fd51b59aefc41819a1194).

This is the error I get from moveit:
```console
[ INFO] [1621454575.663979800, 95.245000000]: Planning attempt 1 of at most 1
[ INFO] [1621454575.666822400, 95.248000000]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1621454575.667156900, 95.248000000]: manipulator/manipulator: Starting planning with 1 states already in datastructure
[ERROR] [1621454580.672333000, 99.602000000]: manipulator/manipulator: Unable to sample any valid states for goal tree
[ INFO] [1621454580.672491800, 99.602000000]: manipulator/manipulator: Created 1 states (1 start + 0 goal)
[ INFO] [1621454580.672859000, 99.602000000]: No solution found after 5.005736 seconds
[ INFO] [1621454580.683008900, 99.607000000]: Unable to solve the planning problem
[ INFO] [1621454580.685616400, 99.609000000]: Received event 'stop'
```

The same moveit commander script works in ROS Melodic (with Python 2.7) and I'm using a similar one for weeks both in simulation and a real UR3e. By the way I have the exactly same issue with the real robot using Noetic, so it's not a Gazebo related problem.

Do you have any idea what can be wrong with MoveIt on Noetic?

Let me know if you have any question, I'm happy to test and/or give further information!

David

fmauch, 2021-05-20:
For reference, my [`calibration_devel-staging`](https://github.com/fmauch/universal_robot/tree/calibration_devel-staging) currently is the same as in #538.

gavanderhoorn, 2021-05-20:
For now I'm going to close this here, but if it turns out something needs to be fixed here, we'll re-open.

closed 2021-05-20 by gavanderhoorn

dudasdavid, 2021-05-20:
It was tested with KDL (since the UR IK doesn't really work well with limited joints and MoveIt). I can partially accept that it's a MoveIt issue, however the same MoveIt Commander application works with Panda robot in the simulation.
I'd also exclude a trivial MoveIt config issue, because I observe the same problem with @fmauch's `calibration_devel` and `calibration_devel-staging` branches.

Is there any planned official Universal Robots MoveIt support, if this repository isn't an official one anymore?

fmauch, 2021-05-20:
> @gavanderhoorn is there any planned official Universal Robots MoveIt support, if this repository isn't an official one anymore?

It's not non-official, it is just not developed / maintained by the Company Universal Robots but driven by community effort.

gavanderhoorn, 2021-05-20:
>  is there any planned official Universal Robots MoveIt support, if this repository isn't an official one anymore?

this may have been unclear, but it's never been "official". It's always been a community effort.

And as I'm not part of UR, I'm not privy to what they plan to do or not.

gavanderhoorn, 2021-05-20:
As to your problem: this will require recreating and then debugging MoveIt. That's unfortunately not something I can make time available for right now.

As a first step, I'd turn up the logging config for MoveIt itself to `DEBUG`, which might provide more insight into *where* exactly something is failing.