ros-industrial/universal_robot #573: this package is outdated with calibration_devel branch

lianghongzhuo, 2021-05-25:
Hi, as you all know, this package is outdated with 100+ commit with https://github.com/fmauch/universal_robot/tree/calibration_devel
This is confusing and breaks a lot of things. Why not just synchronize all commits from Felix?

fmauch, 2021-05-25:
It is not very much out-of-sync. With https://github.com/fmauch/universal_robot/pull/10 the only effective differences are #562 and #538. I decided to do https://github.com/fmauch/universal_robot/pull/10 in order to make my fork more exchangeable with the upstream repo in case someone doesn't want to use MoveIt!. The history looks different, as my [calibration_devel](https://github.com/fmauch/universal_robot/tree/calibration_devel) branch took a different path in order to make users able to always do a `git pull` to get the most recent state. In parallel we worked on getting the changes into this repository, what we already did to a great extent (see #371 for example).

For example, the driver's CI pipeline already  [uses](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/.melodic.rosinstall) the version from this repo.

For everything else, the mentioned PRs are in place. In open source software development things sometimes take a bit more time, as there are always people necessary to critically review any incoming suggestions and changes which is often done aside any day jobs.

While I understand that the current situation between the repositories is not the easiest to understand and I also wish that things get merged quite soon, I think we are in a very good situation already. I think it is [clearly documented](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building) that my fork is currently to be used for the driver. As soon as we can swap back to the upstream version, the driver's documentation will be updated and my fork will not have to be used anymore...

lianghongzhuo, 2021-05-25:
I know you did a good job on maintaining the ur driver and ur descriptions, thank you for that.
But why ros-indestrial maintainers did not react?

lianghongzhuo, 2021-08-06:
Please at least claim this repo as unmaintained since it's outdated and people getting errors when jumping into this repo (https://github.com/ros-industrial/universal_robot/issues/494) while there exist a bug-fixed version (https://github.com/fmauch/universal_robot/tree/calibration_devel) of this repo but not getting merged. I don't know why.

fmauch, 2021-09-29:
@ipa-nhg This could be closed as soon as #538 gets merged, I guess.

lianghongzhuo, 2021-09-29:
Thanks a lot for your efforts to fix this!

ipa-nhg, 2021-09-29:
@fmauch yes, I think so. I linked the issue to get this one automatically closed once is merged

fmauch, 2022-10-14:
This issue can be closed by now.

closed 2022-10-14 by fmauch