ros-industrial/universal_robot #581: `ee_link` exists in SRDF files but not URDF

danielcranston, 2021-09-24:
Looking at the `melodic-devel-staging` branch which I believe to be the main melodic (and noetic?) branch after reading around a bit (and considering it's targeted as the dependant branch in the [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/v2.0.0/.melodic.rosinstall) repo).

I'm curious, what's the deal with `ee_link` ? It is expected in the SRDF files (see e.g. [here](https://github.com/ros-industrial/universal_robot/blob/melodic-devel-staging/ur5_e_moveit_config/config/ur5e.srdf#L13)), but the URDF makes no mention of it (see [here](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging/ur_description/urdf)).

Is the intention to keep `ee_link` in the TF tree or remove it?

gavanderhoorn, 2021-09-24:
`ee_link` has already been removed, and the remnant in the MoveIt configurations will also be removed (in fact: they must be completely recreated).

It never aligned with any physical or conceptual link in the real robot or its control system, so its use was limited.

`tool0` is aligned with an unconfigured tool frame on the UR CB.

The `flange` frame should be used to attach EEF models to.

See [Coordinate Frames for Serial Industrial Manipulators](https://gavanderhoorn.github.io/rep/rep-0199.html) for the guiding principles used to align the UR `xacro:macro` with other robot support packages.

gavanderhoorn, 2021-09-24:
As this is more of a usage question, and not a problem with the packages in this repository, I'm going to close the issue.

Feel free to keep commenting on it of course.

closed 2021-09-24 by gavanderhoorn