ros-industrial/universal_robot #585: description: Should the load_urXX.launch files use the robot-specific macros?

fmauch, 2021-09-29:
As risen in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/471 the robot model loaded by the `load_urXX.launch` files of `ur_description` create a robot called `ur`:

https://github.com/ros-industrial/universal_robot/blob/0101cae607c9a8e7fc72934849eb4f2343e58ded/ur_description/urdf/ur.xacro#L2

When used with the moveit_config packages (see #538), this name clashes with the SRDF files configured there, when the moveit startup should not re-load the description. It might be a good idea to use the named macros inside the `load_urXX.launch` files.

Is there anything speaking against this?

RobertWilbrandt, 2022-11-30:
I cannot quite track the intention mentioned in the title, but is this still relevant? As of #538 MoveIt by default doesn't load its own description anymore and #615 made sure that the srdf robot name matches the urdf.

fmauch, 2023-01-03:
You're right, this should have been resolved by now.

PR #588 (2021-09-29): ur_description: Use the per-model xacro files for correct robot name
With the common xacro file the robot name will always be "ur".
However, the moveit config packages expect the robot name to be e.g. "ur10e".
This commit will use the macros containing the model name already.

This fixes #585 and should help in #538.
closed 2023-01-03 by fmauch