ros-industrial/universal_robot #597: "stack smashing detected" error when generating collision matrix with URDF which contains colliding link geometries

akmandor, 2022-04-24:
Issue:
When generating collision matrix in MoveIt Setup Assistant using the URDF which contains colliding link geometries, the "stack smashing detected" error happens and the Setup Assistant terminates. It would be really helpful if you could update the URDF files to resolve the issue.

Tests:
- The issue is experienced when both creating new or editing the existing MoveIt configuration packages.
- The issue is not experienced when the link geometries in URDF do not collide. For example the default URDF of [jackal robot](https://github.com/jackal/jackal/blob/noetic-devel/jackal_description/urdf/jackal.urdf.xacro) have the same issue but when the "track" property is changed from "0.37559" to "0.38" (which slightly moves the tire way from the main chassis), the collision matrix can be successfully generated using the Setup Assistant.
- The issue is not experienced using the (unofficial) URDF of another robot arm, Franka Emika's Panda robot](https://github.com/justagist/franka_panda_description/blob/master/robots/panda_arm.urdf.xacro). However, when the official URDF of the same robot arm is used in [franka_ros](https://github.com/frankaemika/franka_ros/blob/develop/franka_description/robots/panda_arm.urdf.xacro), the same error is happened.

System details: 
- Ubuntu 20.04 with ROS Noetic (desktop full) installation
- universal_robot branches: kinetic-devel and melodic-devel
- URDF files: ur{3,5,10}{-.e}_robot.urdf.xacro
- MoveIt configs: ur{3,5,10}{-,_e}_moveit_config

gavanderhoorn, 2022-04-25:
While I thank you for the (more) detailed report, I still believe this is a problem with the MSA, not with the URDF here. It's possible #527 and #570 are other instances of this same problem.

The URDF specs don't state that collision geometry cannot overlap, and due to the fact that yours is only the second (or third) report about this, I'm inclined to believe (at this point) it's not a widespread / fundamental problem with the models here.

I would recommend to open an issue on the MoveIt issue tracker, and providing them with as much information as possible. You probably have a `gdb` backtrace, which I would include.

If, after analysis there, it turns out there is a problem with the models in `ur_description` here, we can try to address it of course.

gavanderhoorn, 2022-04-25:
For now I'm going to close this, but as I wrote in my previous comment, we can certainly come back to this and re-open if/when needed.

closed 2022-04-25 by gavanderhoorn

akmandor, 2022-04-25:
Thanks @gavanderhoorn and personally I agree with your comment that "The URDF specs don't state that collision geometry cannot overlap". I think it is more meaningful to expect from the MSA implementation to handle such overlapping geometries. However, in an ideal simulation, I also don't expect the physical entities collapse. Anyway, this is obviously a gray area :) As you suggested, I wrote my observations under a similar [issue](https://github.com/ros-planning/moveit/issues/1596#issuecomment-1108844957).

gavanderhoorn, 2022-04-25:
I would suggest to include a backtrace of an example crash.

I expect that to be one of the first questions you'll receive from the MoveIt maintainers.

akmandor, 2022-04-25:
Added :+1: