ros-industrial/universal_robot #598: Problem with catkin_make

robertokcanale, 2022-05-12:
I tried to run catkin_make

I get this:
Found moveit_core: 2.2.3 (/opt/ros/foxy/share/moveit_core/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
Traceback (most recent call last):
  File "/opt/ros/foxy/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /opt/ros/foxy/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3
  /opt/ros/foxy/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/roberto/Desktop/ROS1_UR/build/universal_robot/ur_kinematics/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:41 (include)
  /opt/ros/foxy/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /opt/ros/foxy/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  universal_robot/ur_kinematics/CMakeLists.txt:6 (find_package)

So apparently it finds some packages that exist in ROS2. How to fix this??

gavanderhoorn, 2022-05-12:
Make sure to build in a clean shell.

Check you're not auto-`source`-ing `/opt/ros/foxy/setup.bash` for instance.

Note that this is not really a question specific to the packages in this repository. It's more about general ROS/ROS2 usage.

gavanderhoorn, 2022-05-12:
As it isn't a problem with any of the packages in this repository, I'm going to close the issue.

Feel free to keep commenting on it, although keep in mind what you report is not specific to the packages in this repository.