ros-industrial/universal_robot #604: [ERROR] [1658975665.993442682]: RRTConnect: Motion planning start tree could not be initialized!

gyt290811, 2022-07-28:
I got this issue：
**[ WARN] [1658995780.555897234]: RRTConnect: Skipping invalid start state (invalid bounds)
[ERROR] [1658995780.555934424]: RRTConnect: Motion planning start tree could not be initialized!
[ERROR] [1658995780.555990100]: RRTConnect: Motion planning start tree could not be initialized!
.**
when I run this package **universal_robot**

how to solve it?

fmauch, 2022-08-01:
Thank you for reporting your issue. In order to help you resolve this, we'll need some additional information:

- What where the **exact commands** that you have run to get everything running?
- What is the **complete shell output** of all the nodes related (Driver, moveit)?
- What are the joint positions your robot is currently located when this error pops up? (The output of `rostopic echo -n1 /joint_states` should help there)

gyt290811, 2022-08-01:
> Thank you for reporting your issue. In order to help you resolve this, we'll need some additional information:
>
> * What where the **exact commands** that you have run to get everything running?
> * [...]

exact commands are:
`roslaunch ur_robot_driver ur5_bringup.launch limited:=true robot_ip:=192.168.1.27 use_lowbandwidth_trajectory_follower:=true`

fmauch, 2022-08-15:
Given the files you posted, you seem to not be using ur_robot_driver with the melodic-devel-staging branch, but you seem to use ur_modern_driver. I'd recommend not using ur_modern_driver unless you have a CB1 or CB2 robot.

Since you are running that setup and start everything with `joint_limited:=true`, ROS and in particular MoveIt! will expect your robot operates in the range `{-pi, pi}` for each joint. Clearly, from the joint values you posted, joints 3 and 5 (You cut off the names, so I can only use the indices to reference them) are outside this limit. Once you (manually) move back the robot's joints to the specified interval, things should hopefully work. 

However, as stated above, I suggest you try using the ur_robot_driver ith the melodic-devel-staging branch from this repo instead. This will also be able to handle joint values outside `{-pi,pi}` interval. Alternatively, you could try **not** passing `joint_limited:=true` to all launch files if you expect the robot to operate outside joint limitations.