ros-industrial/universal_robot #605: Add UR20 support

fmauch, 2022-08-01:
As the [UR20](https://www.universal-robots.com/products/ur20-robot/) has been announced, we should add its description. There are still some (internal) things blocking us from adding it, this issue is meant as a single point of reference whenever the question pops up.

garyraise, 2023-01-20:
We're hoping to start using the UR20 later this year in our projects. Do you think that support will be added soon?

LoyVanBeek, 2023-02-13:
What is the status here? 
Is there work-in-progress that I could continue with perhaps? Or otherwise: how to best get started creating the URDF? 

I can probably get the kinematics from the datasheet, but not the meshes of course.

RobertWilbrandt, 2023-02-13:
Sorry for the late response to the questions. The meshes for the UR20 are not yet publicly available, as soon as they are we will provide patches to add support for them.

LoyVanBeek, 2023-02-13:
Awesome! And thanks for the quick response to my question :smiley: 

Would it make sense to leave the meshes out on some WIP branch/PR and only have the kinematics in a URDF?

Also: I guess I shouln't hold my breath for those meshes, but is there any estimate on when they'll be available? I'd expect UR to have these soon so people can start trying their stuff out in preparation of buying them, but then they would have been available directly after announcement...

urrsk, 2023-03-22:
We are working on making this possible. The goal is to have it available in Q3

LoyVanBeek, 2023-03-22:
Alright, thanks for the update!

fmessmer, 2023-09-13:
obviously, urdf and meshes are now available via https://github.com/ros-industrial/universal_robot/pull/657

urrsk, 2023-09-15:
Yes we are getting closer on the ROS side
Everything is now in place in https://github.com/UniversalRobots/Universal_Robots_ROS2_Description and there is a PR for the ROS2 driver as well

fmauch, 2023-12-18:
Done with #657

closed 2023-12-18 by fmauch

fmessmer, 2023-12-18:
will there be a new release to make the ur20 support available via binaries?

fmessmer, 2023-12-18:
> will there be a new release to make the ur20 support available via binaries?

thanks @fmauch 
https://github.com/ros/rosdistro/pull/39253