ros-industrial/universal_robot #613: Future of ur_kinematics plugin

fmauch, 2022-11-07:
Currently, the main thing holding us back from doing a binary release is the ur_kinematics plugin. The current identified problems are:

* It uses a deprecated API (See #476), while #476 makes the move_group crash on startup
* URKinematics currently only supports UR3, UR5, UR10, while all the four e-Series robots are missing
* The URKinematics plugin uses the uncalibrated robot model, so it won't produce accurate results

I am not sure what the scope / impact of this would be. With many alternative kinematics plugins available, I'm asking myself whether it would be worthwile addressing this. For the sake of no longer holding things back I would currently vote for merging melodic-devel-staging into melodic-devel without fixing that package and ignoring it in the first melodic/noetic release.

If you think that it would be beneficial having this available or you even would like to contribute to fixing this, please let us know.

fmauch, 2022-11-10:
Update: We've updated the package so that it supports all 7 robot models, but the other two points remain valid in my opinion.

simonschmeisser, 2023-07-17:
Just as a quick update, I opened a PR to fix the crash in #476  (I could also open a new MR directly)

For us the analytical IK is valuable as it tends to be about a factor 1000 faster than the numerical ones. Since we can easily have 50 000 IK requests in our bin picking setups this adds up