ros-industrial/universal_robot #625: Semantic description is not specified for the same robot as the URDF

Akumar201, 2022-11-21:
I am trying to launch 
`universal_robot/ur5e_moveit_config/launch$ roslaunch moveit_rviz.launch `
however I am getting the following error, from my understanding there is something wrong with launching move_group.launch, but even after launching this I am getting the following error, 
`Semantic description is not specified for the same robot as the URDF`
After launching move_group.launch I am getting the following error
` Semantic description is not specified for the same robot as the URDF`

RobertWilbrandt, 2022-11-23:
Can you confirm that you use an up-to-date version of this repository? From my understanding this error should be fixed since #615 and i am currently not able to reproduce the error you are describing.

Could you also show your complete sequence of operations that leads to this error? I.e. how exactly did you start gazebo / the robot driver and how did you start the move group?

Akumar201, 2022-11-24:
Hi, I am on the melodic branch. Actually I am using ROS Noetic and was following the tutorial https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html , however it's need to be updated. While attempting to reproduce the tutorial I found this error. Please guide me through it. I am using Ubuntu 20.04 and ROS Noetic

VJan2022, 2022-11-24:
I tried to $roslaunch ur5_moveit_config moveit_rviz.launch
But it happens to 
[ERROR] [1669274019.893303177]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1669274019.894229168]: Robot model not loaded
WHY?

VJan2022, 2022-11-24:
> @VJan2022 please do not hijack other issues. Your error output is clearly different and looks like you didn't start the `roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch` command beforehand.
> 
> @Akumar201 The exact commands that you used to start your system (`roslaunch ....`, together with the output) would be important to find out what is going wrong.

sorry and thank you

Akumar201, 2022-11-25:
I was able to resolve this issue. I build catkin workspace and source my file and then it was working perfectly. The problem was associated with not able to find URDF file from the folder. 

I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2022-11-25 by Akumar201