ros-industrial/universal_robot #626: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint

Akumar201, 2022-12-05:
Hi, I am running gazebo with ur5 using the following command 
`roslaunch ur_gazebo ur5.launch`

I am getting the following error 

```
[ERROR] [1670278868.694721173, 0.377000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint
[ERROR] [1670278868.695056205, 0.377000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint
[ERROR] [1670278868.695361019, 0.377000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint
[ERROR] [1670278868.695658561, 0.377000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
[ERROR] [1670278868.695953383, 0.377000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint
[ERROR] [1670278868.696248616, 0.377000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint

```

I am using Ubuntu 20.04, ROS Noetic and  I checkout to 1.2.7 version.

fmauch, 2022-12-07:
This is a known kind-of non-issue, see also #443 

The error output should not affect the simulation.

Akumar201, 2022-12-09:
This issue has been resolved, if any problem persist feel free to reopen the issue. Thank You

closed 2022-12-09 by Akumar201