ros-industrial/universal_robot #627: Gazebo: Arm falls on start after switch to Effort Controllers

ian-chuang, 2022-12-27:
When launching ur5e_bringup.launch on ur_gazebo on branch melodic-devel-staging, the robot immediately falls when Gazebo starts. The arm can be controlled just fine but needs to be brought back up from where it fell. I think this is due to the change to using effort controllers instead of position controllers. When using position controllers (without setting pids in gazebo_ros_control), the robot will start at its designated initial position. However effort controllers are needed for doing realistic grasps. Is there a way to fix this?

tommyvtran97, 2022-12-27:
I agree with the original post. I noticed the same issues when using the UR launch files.

fmauch, 2023-01-05:
We'll need to look into that...

stevensu1838, 2023-02-27:
same issue here.

$ roslaunch ur_gazebo ur5e_bringup.launch
$ roslaunch ur5e_moveit_config moveit_planning_execution.launch sim:=true
$ roslaunch ur5e_moveit_config moveit_rviz.launch

The robot arm starts at a weird pose and moveit doesn't generate proper movements. Cheers

[image]

job-1994, 2023-08-02:
Hello,

I am having the same issue running on Ubuntu20.04 with Noetic and the universal_robot packages installed via apt (v1.3.1).

The workaround does work, but fixing the launch would be prefereable, is there any update on this?

Would be glad to help set this up if theres any ideas on where to start?