ros-industrial/universal_robot #631: What is the correct way to attach UR5e to Husky?

AldrB, 2023-02-21:
Melodic ROS, Gazebo 9, Ubuntu 18.04. 

I need to attach the UR5e to Husky. I've installed [universal_robot](https://github.com/ros-industrial/universal_robot) and [Husky](https://www.clearpathrobotics.com/assets/guides/melodic/ros/Drive%20a%20Husky.html) packages. Independently, they work fine. I'm having trouble attaching UR5e to Husky because the UR5e files are arranged differently. They require specific parameters to be passed from the launch file (i.e `joint_limited` and `prefix`). Additionally, I noticed that UR5e's urdf file contains a base_link. Husky's base has the same name. This is my attempt to resolve the issue but I get errors. Inside `husky.urdf.xacro`, I've placed these lines at the bottom 

```
    <!-- common stuff -->
    <xacro:include filename="$(find ur_e_description)/urdf/common.gazebo.xacro" />
    <xacro:include filename="$(find ur_e_description)/urdf/ur5e.urdf.xacro"/>
    
    <!-- ur5e -->
    <xacro:ur5e_robot joint_limited="false"/>
    <xacro:ur5e_robot prefix="ur_arm_" />

    <xacro:arg name="arm_xyz" default="$(optenv HUSKY_UR_XYZ -0.105 0.0 0.0)"/>
    <xacro:arg name="arm_rpy" default="$(optenv HUSKY_UR_RPY 0 0 3.1415927)"/>
  
    <joint name="arm_mount_joint" type="fixed">
      <parent link="top_plate_front_link" />
      <child link="ur_arm_base_link" />
      <origin xyz="$(arg arm_xyz)" rpy="$(arg arm_rpy)"/>
    </joint>
```

I get this error `Undefined parameters [prefix]`. When I open `ur5e.urdf.xacro`, I see 

```
  <xacro:macro name="ur5e_robot" params="prefix joint_limited
    shoulder_pan_lower_limit:=${-pi}    shoulder_pan_upper_limit:=${pi}
    shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
    elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
    wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
    wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
    wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}
    transmission_hw_interface:=hardware_interface/PositionJointInterface
    safety_limits:=false safety_pos_margin:=0.15
    safety_k_position:=20">
```

All links depend on this prefix. For example, `<link name="${prefix}base_link" >`. How can I correctly pass these parameters?

closed 2023-02-21 by AldrB

gavanderhoorn, 2023-02-22:
You've already closed this, but just to draw your attention to it: with this:

```xml
    <xacro:ur5e_robot joint_limited="false"/>
    <xacro:ur5e_robot prefix="ur_arm_" />
```

you call the `ur5e_robot` macro *twice*. Unless you want two UR5es, it should probably be:

```xml
    <xacro:ur5e_robot joint_limited="false" prefix="ur_arm_" />
```

AldrB, 2023-02-22:
> You've already closed this, but just to draw your attention to it: with this:
> 
> ```
>     <xacro:ur5e_robot joint_limited="false"/>
>     <xacro:ur5e_robot prefix="ur_arm_" />
> ```
> 
> you call the `ur5e_robot` macro _twice_. Unless you want two UR5es, it should probably be:
> 
> ```
>     <xacro:ur5e_robot joint_limited="false" prefix="ur_arm_" />
> ```

@gavanderhoorn thank you so much. I did actually solve the problem and I do appreciate your feedback.