ros-industrial/universal_robot #637: How to get the raw value of force_torque_sensor?

KentoYamamotoNU, 2023-05-11:
I am using a robot hand (2F-85) from robotiq attached to UR. When UR is started, the value of force_torque_sensor is initialized and the force shown in the /wrench topic is the picture below. [image]
The posture at this time is the picture below. 
[image]
The value of force_sensor is corrected and the value changes as shown in the picture below about 3 minutes later, even though UR is not moved. 
[image]
Also, if the robot hand was rotated 90 degrees around the x-axis and the y and z values will be different, as shown in the picture below. 
[image]
[image]
How can I get the raw data of force_torque_sensor without any correction?

gavanderhoorn, 2023-05-11:
I would assume you are using either [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) or the ROS 2 version of that driver.

In any case I'd suggest to post your question there, as we don't support that driver here.

If you do post your question on `UniversalRobots/Universal_Robots_ROS_Driver`: please don't post screenshots/photos of terminals. Just copy-paste the text from the terminal into your question.

closed 2023-05-11 by gavanderhoorn

KentoYamamotoNU, 2023-05-11:
Thanks for the replies and advice. I reposted it in UniversalRobots/Universal_Robots_ROS_Driver.