ros-industrial/universal_robot #638: Controller-related dependencies missing in package.xml

AndyZe, 2023-05-14:
If you are trying to launch in Gazebo but the controllers do not start, this is probably why:

`sudo apt install ros-noetic-gazebo-ros-contro* ros-noetic-effort-controllers ros-noetic-position-controllers`

fmauch, 2023-05-15:
Which `package.xml` are you referring to? The one from [`ur_gazebo`](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur_gazebo/package.xml) does contain the mentioned dependencies:

https://github.com/ros-industrial/universal_robot/blob/6507bb6756d0065d49b76230ff55f875f099a827/ur_gazebo/package.xml#L24-L29

gavanderhoorn, 2023-06-16:
Assuming this has been 'resolved', I'm closing this.

PS: @fmauch: I just noticed that `ur_gazebo` apparently depends on meta-packages (`effort_controllers` & `position_controllers`), which is an anti-pattern.

closed 2023-06-16 by gavanderhoorn