ros-industrial/universal_robot #643: How to NOT display motion planning request robot in rviz?

genkv, 2023-07-11:
Hello, newbie question here. How do I modify the moveit.rviz file so that it only displays 

1. the current representation of the robot
2. the planned trajectory, if successful

By default, rviz provides the two said above, AND the motion planning request robot (the orange one whose end effector that I can drag to a target pose). How do I not display that while keeping the capability to show current robot status and planned trajectory? I tried to delete different elements from the moveit.rviz but it hasn't been successful.

Thank you

fmauch, 2023-07-11:
This is not really a bug report / issue with this repo but a general question. Please ask such questions on https://robotics.stackexchange.com/ in future.

To achieve what you want, you'll have to remove the tick at "Query goal state" in the "Planning Request" part of the "MotionPlanning" Display:
[image]

If you always want to show the currently planned trajectory you might want to have "MotionPlanning->Planned Path->Loop Animation" activated.

I hope, this helps, otherwise please open a topic at https://robotics.stackexchange.com.

closed 2023-07-11 by fmauch

genkv, 2023-07-11:
> This is not really a bug report / issue with this repo but a general question. Please ask such questions on https://robotics.stackexchange.com/ in future.
>
> To achieve what you want, you'll have to [...]

Thank you. I am aware now.