ros-industrial/universal_robot #656: Difference between current of the motor on teach Pendant screen and /joint_states values (Effort)?

MohamedOmar2014, 2023-08-28:
I'm working on a real UR10 robot and I concern measuring the current of motor for further analysis of the motor torqe because i cant get it directly. I noticed that their is difference between current values on teach pendant screen and the values from the joint_states topic for a given configuration of the robot as shown in the following photos.
The current values (joint_states): -4.1294   -9.2880   -0.1740    -0.4531    0.2624    -0.0493 
The current values (Screen):         0.2           9.3          3.9            0.5          0.2           0
It seems that in the joint_states, the current value of joint 1 is replaced with current value of joint 3 because the current value of joint 3 has to be bigger based on the robot configuration.

Is that right or their is something missing?
thanks in advance

[image]
[image]
[image]

RobertWilbrandt, 2023-08-28:
Did you check the order of the joints in the `name` field of the [JointState message](http://docs.ros.org/en/api/sensor_msgs/html/msg/JointState.html)? This should be consistent with the order of the positions/velocities/efforts and you cannot rely on the order of those fields not changing.

MohamedOmar2014, 2023-08-28:
Yes, the order is correct, positions/velocities/efforts. the problem is not the order of the positions/velocities/efforts, the problem is with the order of effort(current) of the joints itself.