ros-industrial/universal_robot #672: No UR5 model in gazebo

Yet-ian, 2024-01-03:
When I roslaunch execute the following command ur_gazebo ur5_bringup.launch, Robotic arm model cannot be displayed in gazebo. But the output in the terminal seems normal.

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /joint_group_pos_controller/joints: ['shoulder_pan_jo...
 * /joint_group_pos_controller/type: position_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 10
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 125.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ros_control_controller_spawner (controller_manager/spawner)
    ros_control_stopped_spawner (controller_manager/spawner)
    spawn_gazebo_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [8558]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6a1edc90-aa15-11ee-ac74-b727aa385e75
process[rosout-1]: started with pid [8568]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [8575]
process[gazebo-3]: started with pid [8576]
process[gazebo_gui-4]: started with pid [8580]
process[spawn_gazebo_model-5]: started with pid [8585]
process[ros_control_controller_spawner-6]: started with pid [8587]
process[ros_control_stopped_spawner-7]: started with pid [8592]
[ INFO] [1704271945.879807302]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1704271945.880620311]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1704271945.933302851]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1704271945.934067506]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1704271946.186580186, 0.001000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1704271946.199761296, 0.014000000]: Physics dynamic reconfigure ready.
[image]

Gaojcc, 2024-04-07:
I have also encountered this problem. Have you resolved it？

RobertWilbrandt, 2024-04-08:
This seems like some dependencies are missing, that is way too little output for the `roslaunch ur_gazebo ur5_bringup.launch` command. Did you install the package from binary packages or from source? And in the second case, did you follow the README instructions and install all dependencies using `rosdep`?

For reference, this is what a complete startup should look like:

```
> $ roslaunch ur_gazebo ur5_bringup.launch
... logging to /home/ubuntu/.ros/log/0794ac64-f5ae-11ee-9181-a76ff8ca333b/roslaunch-ur-noetic-1042.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ur-noetic:39381/

SUMMARY
========

PARAMETERS
 * /eff_joint_traj_controller/action_monitor_rate: 10
 * /eff_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /eff_joint_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /eff_joint_traj_controller/constraints/goal_time: 0.6
 * /eff_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /eff_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
 * /eff_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /eff_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.1
 * /eff_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /eff_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /eff_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.1
 * /eff_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /eff_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.1
 * /eff_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /eff_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.1
 * /eff_joint_traj_controller/gains/elbow_joint/d: 20
 * /eff_joint_traj_controller/gains/elbow_joint/i: 1
 * /eff_joint_traj_controller/gains/elbow_joint/i_clamp: 1
 * /eff_joint_traj_controller/gains/elbow_joint/p: 2000
 * /eff_joint_traj_controller/gains/shoulder_lift_joint/d: 200
 * /eff_joint_traj_controller/gains/shoulder_lift_joint/i: 1
 * /eff_joint_traj_controller/gains/shoulder_lift_joint/i_clamp: 1
 * /eff_joint_traj_controller/gains/shoulder_lift_joint/p: 10000
 * /eff_joint_traj_controller/gains/shoulder_pan_joint/d: 200
 * /eff_joint_traj_controller/gains/shoulder_pan_joint/i: 1
 * /eff_joint_traj_controller/gains/shoulder_pan_joint/i_clamp: 1
 * /eff_joint_traj_controller/gains/shoulder_pan_joint/p: 4000
 * /eff_joint_traj_controller/gains/wrist_1_joint/d: 1
 * /eff_joint_traj_controller/gains/wrist_1_joint/i: 1
 * /eff_joint_traj_controller/gains/wrist_1_joint/i_clamp: 1
 * /eff_joint_traj_controller/gains/wrist_1_joint/p: 500
 * /eff_joint_traj_controller/gains/wrist_2_joint/d: 1
 * /eff_joint_traj_controller/gains/wrist_2_joint/i: 1
 * /eff_joint_traj_controller/gains/wrist_2_joint/i_clamp: 1
 * /eff_joint_traj_controller/gains/wrist_2_joint/p: 500
 * /eff_joint_traj_controller/gains/wrist_3_joint/d: 0.1
 * /eff_joint_traj_controller/gains/wrist_3_joint/i: 0
 * /eff_joint_traj_controller/gains/wrist_3_joint/i_clamp: 1
 * /eff_joint_traj_controller/gains/wrist_3_joint/p: 10
 * /eff_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /eff_joint_traj_controller/state_publish_rate: 125
 * /eff_joint_traj_controller/stop_trajectory_duration: 0.5
 * /eff_joint_traj_controller/type: effort_controller...
 * /gazebo/enable_ros_network: True
 * /joint_group_eff_controller/joints: ['shoulder_pan_jo...
 * /joint_group_eff_controller/type: effort_controller...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 125.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: noetic
 * /rosversion: 1.16.0
[... 29 lines cut ...]
[ INFO] [1712583831.273997909]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1712583831.407910172]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1712583831.409627044]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1712583835.414074272]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1712583835.435852688, 0.010000000]: Physics dynamic reconfigure ready.
[INFO] [1712583835.486458, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1712583835.670876, 0.157000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1712583835.705510690, 0.157000000]: Loading gazebo_ros_control plugin
[ INFO] [1712583835.705683720, 0.157000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1712583835.706807832, 0.157000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1712583835.837844938, 0.157000000]: Loaded gazebo_ros_control.
[INFO] [1712583835.993950, 0.270000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1712583835.996028, 0.273000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1712583835.997725, 0.274000]: Loading controller: joint_state_controller
[INFO] [1712583836.004077, 0.281000]: Loading controller: eff_joint_traj_controller
[INFO] [1712583836.037891, 0.314000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[... cut ...]
[INFO] [1712583836.043677, 0.320000]: Loading controller: joint_group_eff_controller
[INFO] [1712583836.108537, 0.385000]: Controller Spawner: Loaded controllers: joint_state_controller, eff_joint_traj_controller
[INFO] [1712583836.130911, 0.392000]: Controller Spawner: Loaded controllers: joint_group_eff_controller
[INFO] [1712583836.131686, 0.393000]: Started controllers: joint_state_controller, eff_joint_traj_controller
[spawn_gazebo_model-5] process has finished cleanly
log file: /home/ubuntu/.ros/log/0794ac64-f5ae-11ee-9181-a76ff8ca333b/spawn_gazebo_model-5*.log

```

Gaben21, 2024-06-13:
@RobertWilbrandt 
I also met this problem after  " [INFO] [1704271946.199761296, 0.014000000]: Physics dynamic reconfigure ready. "
```
libcurl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to fuel.gazebosim.org:443 
libcurl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to fuel.ignitionrobotics.org:443 
```
I install this package from the github repository ' https://github.com/fmauch/universal_robot '

Gaben21, 2024-06-14:
I first used `sudo apt-get install ros-$ROS_DISTRO-universal-robots`
then I used `roslaunch ur_gazebo ur5_bringup.launch`
but I still encountered this problem
[image]
And the UR5 model not loaded in gazebo

fmauch, 2024-06-19:
@Gaben21 could you maybe provide more output? Did you delete the source-built version from your workspace and cleaned and built your workspace again? Otherwise it might still try to use your old source-built version.

Muhammadjunaidmalik1, 2024-07-02:
> @Gaben21 could you maybe provide more output? Did you delete the source-built version from your workspace and cleaned and built your workspace again? Otherwise it might still try to use your old source-built version.

Hi fmauch ! 

I have tried to run the Gazebo version of repo and I am encountering same issue :

[image]

fmauch, 2024-07-03:
I've just verified that with a clean system things seem to be working. I created a Dockerfile:

```Dockerfile
FROM ros:noetic
RUN addgroup --gid 1000 ros_user && \
    adduser --uid 1000 --gid 1000 --home /home/ros_user --disabled-password --gecos "" ros_user
ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y ros-noetic-ur-gazebo
USER 1000:1000
ENV DISPLAY=:0
```

Assuming here, that your host's user ID is 1000.

then, building an image

```console
docker build -t ur_noetic .
```

and starting it

```
docker run -it --rm --net host -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix --device /dev/dri/card1:/dev/dri/card0 ur_noetic
ros_user@ids-felino:/$ roslaunch ur_gazebo ur5_bringup.launch
```

This brings me a working ur5 gazebo simulation.

Muhammadjunaidmalik1, 2024-07-04:
Thanks fmauch , turns out trying it on my another pc worked just fine for me.

closed 2025-05-06 by urfeex